SerafinTech/ST-node-ethernet-ip #1: Unconnected send timeout value

gfcittolin, 2020-12-09:
Hi there, I'm the maintainer of the Node-RED node for Ethernet/IP. We currently use there the original library by Canaan, but it looks like it's not maintained anymore. I'm considering switching to this fork here.

Taking advantage that we'll be modifying the node, there's a small change that would be nice to have in order to support Omron PLCs that are currently not supported.

The default value of the unconnected send is currently 2000 ms [here](https://github.com/SerafinTech/ST-node-ethernet-ip/blob/6a2e768c7a2b4aa888e324f0459c9355ea7f5d56/src/enip/cip/unconnected-send/index.js#L54), but it needs to be at least 5024 ms for Omron PLCs to accept and process the request.

In my tests, changing it to 5024 doesn't affect the communication to the Rockwell PLCs I've tested, so we could either:
 - just change the default to 5024, or
 - make this value configurable as a constructor parameter of the Controller class

Either way, I can create a PR with the needed changes. What do you think about it?

Thanks!

SerafinTech, 2020-12-09:
If you want to put in a PR, I would prefer that you make it configurable as a constructor parameter. I'll then merge and publish to NPM.

SerafinTech, 2020-12-13:
Version 2.3.1 published to NPM

closed 2020-12-13 by SerafinTech