SerafinTech/ST-node-ethernet-ip #3: Feel free to open a PR on the original repo

cmseaton42, 2021-03-22:
## Quick Question
Do you want to pull your changes to the original repository? Feel free to open a PR if you like. I can make you a maintainer so you pull your own PRs as I rarely have the time to look through code reviews these days. 

Also, I am in the middle of a typescript rewrite of the repository that is much more stable (currently 100% coverage). Feel free to check that guy out as well if you like.  The long term plan (assuming I ever get the time) is to fully transition to typescript.

r4nd0wn, 2022-01-31:
I dont know whats going on with the maintainers. No responds to any request. I have a pull request open which never has been answered. Due to the Licence, feel free to merge it back to your repo :D

r4nd0wn, 2022-02-05:
Any news on that?

balbatross, 2023-02-07:
@cmseaton42  @SerafinTech  As another maintainer of a node-ethernet-ip fork (https://github.com/TheTechCompany/node-ethernet-ip) it'd be great to collaborate on a shared repo. I'll be supporting my fork for the next 10-15 years out of necessity so it'd be great to share the work with the community and get to the bottom of this lib.