SerafinTech/ST-node-ethernet-ip #4: Can't read String

jschenkDD, 2021-04-01:
If i try to read string value i got error: **Unrecognized Type Passed Read from Controller: 208**

My code is:
` const stringStructure = new Structure('A_AUTOMATIC.RunRecipeName', tagList);
        try {
            await PLC.readTag(stringStructure);
            console.log(stringStructure.value)
        } catch(e) {
            console.log(e)
        }`

SerafinTech, 2021-04-09:
What is A_AUTOMATIC ? Is it a UDT or a Program?

What type of Controller are you reading from?

SerafinTech, 2021-08-29:
I believe I fixed that on latest commit.

pilotcam, 2021-09-04:
Latest commit is better - but I tested a bit more this time:

- [ ]  `new Structure('A_AUTOMATIC', tagList);` returns `RunRecipeName:{LEN: 17, DATA: Array(82)}`

- [ ] `new Tag('A_AUTOMATIC.RunRecipeName', null, EthernetIP.CIP.DataTypes.Types.STRING);` returns `Buffer(88)`

- [x] `new Structure('A_AUTOMATIC.RunRecipeName', tagList);` returns `This is my recipe`

I'll fiddle with this a bit more today.

pilotcam, 2021-09-05:
Submit PR #9 .  Resolved the string within a structure (item 1 above).

I had the 'Tag' read working also, but decided to abandon those changes because it's difficult to correctly support a write without the template. 

I also resolved a similar issue with string writes.

SerafinTech, 2021-09-07:
Latest commit made it easier to work with structures (See Code Example) and includes PR #9 

```javascript
const { Controller, Tag, TagList, Structure } = require("st-ethernet-ip");

const PLC = new Controller();

PLC.connect("192.168.1.1", 0).then(async () => {
    
    const tagList = new TagList();
    await PLC.getControllerTagList(tagList);

    //Setup 'Structure1' Tag
    const structure1 = new Structure('Structure1', tagList, {Optional Program Name});
    
    //Read value from PLC
    await PLC.readTag(structure1);

    //Display entire structure
    console.log(structure1.value);

    //Change value of structure member 'Integer1'
    structure1.value.Integer1 = 3;

    //Write Changes to PLC
    await PLC.writeTag(stringStructure)
    
});

```

pilotcam, 2021-09-07:
Oh I like that change.   Certainly addresses a not-so-obvious behavior of the prior value setter.  Thanks!

SerafinTech, 2021-09-08:
Couldn't sleep.  Fixed structure issue with scanning function.  Additional bug fixes.  PR #10

closed 2021-09-08 by SerafinTech