SerafinTech/ST-node-ethernet-ip #5: Ethernet/IP Scanner Function

Venomz101, 2021-04-29:
Hello creator and good work with this repository!!

I have a few questions and I am unsure where to bring them up other than here, so excuse me if this is the wrong forum.

I was thrilled to see that this repository had I/O Scanner as I am currently trying to establish communication to a Fanuc Robot Controller R-30iB Mate. The robot controller works as Ethernet/IP Adapter and writes its inputs to Instance number e.g 101 and outputs to Instance number e.g 151.

The way I would like to achieve communication is through explicit message (tcp) to be able to read both I/Os and register acyclically, cyclic is not important so implicit message (udp) might be good to have for future case.

I am currently looking to try to implement this to Node-RED where configuration can simply be made what instances are to be read and written by specifying:
* Class
* Instance
* Attribute
* Service

Reading the instance numbers e.g 101 and 151 is priority and will get me on the way for what I am currently working on, but having the flexibility to specifiy what to read and write will give me almost complete data exchange with any Ethernet/IP adapter.

Now to the questions,

* Do you think this is possible with the current state of this repository? I noticed in the example code that I/O Scanner was Alpha and not for production.
* Depending on answer above, is there a timeframe when this will be complete? Is it being further worked on/developed?

Attached is a .pdf file for the EIP for the Fanuc Robot Controller, especially interesting is page 92-142, page 139 is good for what I want to achieve foremost.

[R-30iB_EIP_operator_manual_[MAROC77EN01101E_Rev.G].pdf](https://github.com/SerafinTech/ST-node-ethernet-ip/files/6397082/R-30iB_EIP_operator_manual_.MAROC77EN01101E_Rev.G.pdf)

SerafinTech, 2021-04-30:
**USE AT YOUR OWN RISK**
**DO NOT USE TO SECURE SAFETY OF PEOPLE OR PROPERTY**

To answer your questions.

- The I/O scanner is alpha because it has only been tested on a couple devices (Beckhoff ethernet/ip bus coupler and E1 Plus EtherNet/IP Side Mount Module) and not for very long.  It has not been tested in any kind of industrial setting.
- I will be happy to implement any bug fixes or feature additions that may be requested, but this project will never technically be completed.

So your request to be able to read/write any instance you wanted - I have created a fork with that ability.

[cip-functions](https://github.com/SerafinTech/ST-node-ethernet-ip/tree/cip-functions)

Example

```javascript
const { Controller } = require("ST-ethernet-ip")

const controller = new Controller();

// Third parameter is for custom path to controller for devices that are not Allen Bradley PLCs.
// Usually an empty buffer
controller.connect('192.168.86.40', 0, Buffer.alloc(0) ).then(async () => {
  
  // 0x04 - Assembly Class
  // 101 - Instance
  // 3 - Attribute
  const data = await controller.cipGetAttributeSingle(0x04, 101, 3)
  
  console.log(data)

}).catch(e => {
  //Error Handling
  console.log(e)
})
```

**USE AT YOUR OWN RISK**
**DO NOT USE TO SECURE SAFETY OF PEOPLE OR PROPERTY**

Venomz101, 2021-05-03:
This is great! Thank you.

I will test these functions towards the Fanuc R-30iB Mate controller when the opportunity arises and get test results back to you.

closed 2021-08-29 by SerafinTech