SerafinTech/ST-node-ethernet-ip #14: Unable to read tags outside of PLC.connect callback function

skykep, 2021-10-04:
## Current Behavior
`    PLC.connect("192.168.2.235", 1).then(async (err) => {
<read the tag here>
    });`
If the code to read a tag is within that callback, it works fine; however, if it lies within another function, say on an interval, it times out.  Although, I can read PLC.properties just fine in the external functions.

## Expected Behavior
Once the PLC is connected, I should be able to summon PLC.readtag from elsewhere.

## Context
Set up an interval where a tag's value is read and displayed in the console.

## Your Environment

* Package version (Use `npm list` - e.g. *1.0.6*): 2.4.4
* Node Version (Use `node --version` - e.g. *9.8.0*): 12.18.3
* Operating System and version: Ubuntu 18.04
* Controller Type (eg 1756-L83E/B): Tested with both CompactLogix L18 and ControlLogix L73
* Controller Firmware (eg 30.11): 20.12 & 31.11

skykep, 2021-10-20:
That worked, but I don't know why this is any different from the original.  

Similarly, what is the proper way to disconnect a PLC?  For now, I have just been using PLC.pauseScan.

kgustine, 2022-07-28:
I am also having a heck of a time useing `PLC.readTag` and `PLC.writeTag`.  These worked perfectly in the original library.  I am using the `ControllerManager` and return a reference to the original `Controller` object.  I then use that `Controller` object to try and `readTag` and `writeTag`.  `readTag` just hangs up if I use await within an `async` function.  `writeTag` will not set the value if I supply it in the function call like the original library did.  I have to manually set the value before calling `writeTag`.  Even setting it, it still isn't writing to the controller.  I tried it in connected mode and unconnected mode, though I do not understand the difference.  Subscribing to tags works fine and I get those updates.

SerafinTech, 2022-07-28:
Hmm...  If you can post a few snippets of code,  I maybe able to help.  You maybe trying to read or write a tag before the controller connects.  With the ControllerManager you would not usually use writeTag or readTag. You just set the value of the tag and the controller will automatically take care of writing it to the PLC.  You want to make sure the tag is initialized before reading or writing to the tag.

kgustine, 2022-07-28:
The first write works, setting the BOOL to a 1.  The second write within the timeout function fails.

I fell back to not using the controller manager just to see if I could get it to work like the original library.

```
const {Controller, Tag} = require('st-ethernet-ip')

const PLC = new Controller();

PLC.connect('10.*.*.*', 0, 1000, false).then(async () => {
    testTag = new Tag('test_bool', null, 193)
    testTag.value = 1
    PLC.writeTag(testTag).catch(err => { console.log('Error Writing Tag', err) })
})

setTimeout(() => {
    testTag.value = 0
    PLC.writeTag(testTag).catch(err => { console.log('Error on Write', err) })
}, 2000)
```

`Error on Write Error: TIMEOUT occurred while writing Writing Tag: test_bool.
    at Controller._writeTag (D:\NodeJS\tagWrite\node_modules\st-ethernet-ip\src\controller\index.js:850:29)
    at TaskEasy._runTask (D:\NodeJS\tagWrite\node_modules\task-easy\src\index.js:118:9)
    at TaskEasy._next (D:\NodeJS\tagWrite\node_modules\task-easy\src\index.js:138:18)
    at D:\NodeJS\tagWrite\node_modules\task-easy\src\index.js:56:22
    at new Promise (<anonymous>)
    at TaskEasy.schedule (D:\NodeJS\tagWrite\node_modules\task-easy\src\index.js:51:16)
    at Controller.writeTag (D:\NodeJS\tagWrite\node_modules\st-ethernet-ip\src\controller\index.js:609:35)
    at Timeout._onTimeout (D:\NodeJS\tagWrite\src\fpis\plc.js:22:9)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)`

SerafinTech, 2022-07-28:
Change `const PLC = new Controller();` to  `const PLC = new Controller(false);`
Change `PLC.connect('10.163.86.42', 0, 1000, false).then(async () => {` to `PLC.connect('10.163.86.42', 0).then(async () => {`
Change `}, 2000)` to `}, 5000)`

And give it a try

kgustine, 2022-07-28:
Your first reply gave me the clue I needed.  I believe one of your core changes is that you read the entire tag list out of the PLC, which requires longer to connect.  At this point I can't say what I have and haven't tried, but here is my working code.  Thanks for the clues!!

```
const {ControllerManager} = require('st-ethernet-ip')

//Connection Manager automagically reconnects to controller that have lost connection. Scanner auto initiated.
const cm = new ControllerManager();

//addController(ipAddress, slot = 0, rpi = 100, connected = true, retrySP = 3000, opts = {})
const PLC = cm.addController('10.163.86.42', 0, 1000);

PLC.connect();

PLC.on('Connected', () => {
    PLC.addTag('tag1')
    PLC.addTag('tag2')
    PLC.addTag('tag3','program2')
    testBool = PLC.addTag('test_bool')
})
```

I am then able to write to the value of testBool at will.

SerafinTech, 2022-07-29:
Glad I could help =)

closed 2023-08-09 by SerafinTech