SerafinTech/ST-node-ethernet-ip #17: issues with safety plc

r4nd0wn, 2022-02-07:
cannot connect to safety controller

## Current Behavior
when I try to connect to a normal PLC, there are no issues. If I try to connect to a safety PLC (1756-L61S) I cannot establish a connection.

## Expected Behavior
successfully connect without timeout.

## Possible Solution (Optional)
I will have to dive deeper into the code, to check the connection, perhaps I am able to fix it myself.

## Steps to Reproduce (for bugs only)
// connect with a safety in slot 7
const {ControllerManager} = require('st-ethernet-ip')
const cm = new ControllerManager();

const cont = cm.addController('192.168.20.3', slot=7);

cont.connect();

cont.addTag('testtag')

cont.on('TagChanged', (tag, prevValue) => {
  console.log(tag.name, ' changed from ', prevValue, ' => ', tag.value)
})

cont.on('error', (e) => {
  console.log(e)
})

## Your Environment
* Package version (Use `npm list` - e.g. *1.0.6*): 2.4.4
* Node Version (Use `node --version` - e.g. *9.8.0*): latest LTS
* Operating System and version: debian buster
* Controller Type (eg 1756-L83E/B): 1756-L61S
* Controller Firmware (eg 30.11): no idea

r4nd0wn, 2022-02-07:
currently just for issue tracking, I have to dive deeper into that issue myself.

SerafinTech, 2022-11-12:
Should be fixed see #21

closed 2023-08-09 by SerafinTech