SerafinTech/ST-node-ethernet-ip #18: why doesn't await PLC.getControllerTagList(tagList) work?

planetixin, 2022-04-05:
## Current Behavior

returns:
node:436) UnhandledPromiseRejectionWarning: #<Object>
(Use `node --trace-warnings ...` to show where the warning was created)
(node:436) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict`
(see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:436) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

## Expected Behavior

it should get tag list from PLC.

## Possible Solution (Optional)

## Context

const {Controller, Tag, TagList, TagGroup, Structure} = require('ethernet-ip');

const PLC = new Controller();
const tagList = new TagList();

PLC.connect('address ip', 0).then(async () => {
    await PLC.getControllerTagList(tagList);
    console.log(tagList);

}); 

## Steps to Reproduce (for bugs only)

1.
2.
3.
4.

## Your Environment

* Package version (Use `npm list` - e.g. *1.0.6*):
* Node Version (Use `node --version` - e.g. *9.8.0*):
* Operating System and version: windows 7
* Controller Type (eg 1756-L83E/B):
* Controller Firmware (eg 30.11):

SerafinTech, 2022-04-05:
Should be `require('st-ethernet-ip')` .To show the actual error add a `catch()` to the end of the `PLC.connect()` function.  reply back what you get.

```
const {Controller, Tag, TagList} = require('st-ethernet-ip');

const PLC = new Controller();
const tagList = new TagList();

PLC.connect('address ip', 0).then(async () => {
await PLC.getControllerTagList(tagList);
console.log(tagList);

}).catch(e => { console.log(e) } );
```

planetixin, 2022-04-06:
something like that? { generalStatusCode: 5, extendedStatus: [ 0 ] }

planetixin, 2022-04-06:
OK now it says: [object Object].

SerafinTech, 2022-04-06:
That error is `Request Path destination unknown: Probably instance number is not present`

Are you connected to the ethernet port on the controller or a card plugged into a slot? 

Also double check IP address and check controller properties on the security tab that you don't have the communications restricted.

planetixin, 2022-04-07:
I pinged that address ip and everything is fine. PLC model is Intel(R) 82579LM Gigabit Network Connection.

SerafinTech, 2022-04-07:
PLC models that this library is compatible with are Compactlogix and Controllogix from Allen Bradley.  I believe  Intel(R) 82579LM Gigabit Network Connection is not a PLC