SerafinTech/ST-node-ethernet-ip #19: having issues running sample codes with UnhandledPromiseRejection

kevinjjjjj, 2022-06-01:
I am trying to run a sample code under the heading 'Getting a List of Available Controller Tags and Structure Templates'

but I keep into running this error upon execution of this line:

PLC.connect("192.168.1.1", 0).then(async () => {...}

Uncaught Error UnhandledPromiseRejection: This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason "#<Object>".
    at getErrorWithoutStack (undefined:320:15)
    at generateUnhandledRejectionError (undefined:339:15)
    at processPromiseRejections (undefined:286:24)
    at processTicksAndRejections (undefined:96:32)

## Current Behavior
keep into running this error:
Uncaught Error UnhandledPromiseRejection: This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason "#<Object>".
    at getErrorWithoutStack (undefined:320:15)
    at generateUnhandledRejectionError (undefined:339:15)
    at processPromiseRejections (undefined:286:24)
    at processTicksAndRejections (undefined:96:32)

upon execution of this line:
PLC.connect("192.168.1.1", 0).then(async () => {...}

## Expected Behavior
return of tag list

## Steps to Reproduce (for bugs only)
1. Run the following code:
const { Controller, TagList } = require("st-ethernet-ip");

const PLC = new Controller();

const tagList = new TagList();

PLC.connect("192.168.1.1", 0).then(async () => {
    
    // Get all controller tags and program tags
    await PLC.getControllerTagList(tagList)

    // Displays all tags
    console.log(tagList.tags)

    // Displays all templates
    console.log(tagList.templates)

    // Displays program names
    console.log(tagList.programs)

    
});

## Your Environment

* Package version (Use `npm list` - e.g. *1.0.6*):
├── ethernet-ip@1.2.5
└── st-ethernet-ip@2.4.5
* Node Version (Use `node --version` - e.g. *9.8.0*):
v18.2.0
* Operating System and version:
windows 10
* Controller Type (eg 1756-L83E/B):
omron nx1p2
* Controller Firmware (eg 30.11):
1.27

SerafinTech, 2022-06-02:
Try this first to see if you can see PLC properties and catch errors
```javascript
const { Controller, TagList } = require("st-ethernet-ip");
const PLC = new Controller();

const tagList = new TagList();

PLC.connect("192.168.1.1", 0).then(async () => {

console.log(PLC.properties);

}).catch( e => {console.log(e)} );
```

Also try unconnected messaging by changing

```javascript
const PLC = new Controller();
```

To

```javascript
const PLC = new Controller(false);
```

kevinjjjjj, 2022-06-02:
Thank you very much for your assistance. 

I have followed your suggestion and it looks like the code runs without raising any error but I am getting this message in the console:

`{generalStatusCode: 8, extendedStatus: Array(0)}`

and frankly I am not sure what the code 8 means. Is there a document that I can refer to see what general status code 8 means?

SerafinTech, 2022-06-02:
The service code stands for: Service Not Supported

That PLC does not support retrieving a taglist from the controller (Class 0x6b).  The current version of this library automatically grabs a taglist from the controller when you connect, which is why the error appears. I will look at adding the option to not grab the taglist when connecting to PLC in a future version, but that still will not allow you to see the tag names.

Reference:
[w506_nx_nj-series_cpu_unit_built-in_ethernet_ip_port_users_manual_en.pdf](https://assets.omron.eu/downloads/manual/en/v2/w506_nx_nj-series_cpu_unit_built-in_ethernet_ip_port_users_manual_en.pdf)

kevinjjjjj, 2022-06-02:
oh wow, thank you very much for the reference material! 

I will try other functions and see if I can get omron plc to work with your api.

closed 2023-08-09 by SerafinTech