SerafinTech/ST-node-ethernet-ip #21: Cannot connect to PLC. General Status Code: 5

petrsysel, 2022-07-12:
## Current Behavior
I'm trying connect to PLC with this code:
```js
const {Controller} = require('st-ethernet-ip')

const PLC = new Controller();

PLC.connect("192.168.1.222", 0).then(async () => {

    console.log("Connected!")
    
}).catch(async e => {
    console.log(e)
});
```

And I get this in the console:
```js
{ generalStatusCode: 5, extendedStatus: [ 0 ] }
```

I read here: [https://assets.omron.eu/downloads/manual/en/v2/w506_nx_nj-series_cpu_unit_built-in_ethernet_ip_port_users_manual_en.pdf](url) that is a "Path destination unknown" error.

When I replace library "st-ethernet-ip" with "ethernet-ip", everything works ok. But I need this library, because of UDTs.

Can someone help?

Thank you

## Expected Behavior
Connect to PLC

## Steps to Reproduce
Run the code see above

## Environment
* Package version:
  * ethernet-ip@1.2.5
  * st-ethernet-ip@2.4.7
* Node Version: v16.15.1
* Operating System and version: Windows 11
* Controller Type: Allen Bradley 5069-L330ERMS2
* Controller Firmware: 34.011

petrsysel, 2022-07-15:
Thank you very much for your answer.

I tried unconnected messaging, but nothing changed. I still can't connect. Is there any other way to fix this?

Thank you

SerafinTech, 2022-07-18:
I'm going to make a change so that we can know where in the program we are actually getting the error and maybe some additional meta data in the error report.  Probably by this evening.

SerafinTech, 2022-07-19:
Try to installing from the GitHub repository and running again.  The error report should have some extra data to help see where the error is happening.

`npm install SerafinTech/ST-node-ethernet-ip`

petrsysel, 2022-08-01:
Thank you very much for your assistance.
Unfortunately, I don't have the hardware available right now and I don't know, when I will be able to try this.

ReisoAphi, 2022-09-07:
I'm having the same issue specific with 1756-L81ES PLC, any advice?

SerafinTech, 2022-09-10:
Safety controllers are having an issue.  I don't currently have access to one to troubleshoot.

srebling, 2022-11-02:
I am experiencing a similar error when connecting to a non-safety CompactLogix (1769-L24ER-QB1B 5730 Controller Firmware 30.011) . I can also replace st-ethernet-ip with ethernet-ip and it works but I need it for the UDT's as well.  Any thoughts? 

Error object returned from catch:
```
{
  generalStatusCode: 5,
  extendedStatus: [ 0 ],
  ext: [
    {
      func: '_getTemplateAttributes',
      templateID: 193,
      cipReq: <Buffer 03 02 20 6c 24 c1 04 00 04 00 05 00 02 00 01 00>,
      attributes: [Object],
      members: [],
      name: ''
    }
  ]
}
```

SerafinTech, 2022-11-02:
I published a branch that should skip the getTemplateAttributes that is giving you the issue.

`npm install SerafinTech/ST-node-ethernet-ip#issue21`

please give it a try and feedback would be appreciated so I can merge it to the main branch

srebling, 2022-11-03:
Thanks for looking at it so quickly. That worked. What is the issue with getTemplateAttributes in my case?

ReisoAphi, 2022-11-03:
> I published a branch that should skip the getTemplateAttributes that is giving you the issue.
> 
> `npm install SerafinTech/ST-node-ethernet-ip#issue21`
> 
> please give it a try and feedback would be appreciated so I can merge it to the main branch

Thanks sir, It also worked for me. on a safety PLC "1756-L81ES"

SerafinTech, 2022-11-04:
Some PLCs must have tags with templates that your not allowed to read the attributes of.  Very strange. Probably some special system tags/templates that are restricted.

SerafinTech, 2022-11-04:
Merged and published update to NPM

closed 2023-08-09 by SerafinTech