SerafinTech/ST-node-ethernet-ip #22: What does "status" means in properties object?

petrsysel, 2022-08-02:
I  need to know, what exactly does "status" means in propetries object. Example:
```js
{
  name: '5069-L340ERM/A',
  serial_number: 1620037161,
  slot: 0,
  time: null,
  path: <Buffer 01 00>,
  version: '34.11',
  status: 8304,
  faulted: false,
  minorRecoverableFault: false,
  minorUnrecoverableFault: false,
  majorRecoverableFault: false,
  majorUnrecoverableFault: false,
  io_faulted: false
}
```
I need it to know if the PLC is in run mode or not. Although I trace which numbers belong to which modes, but I would like to know the full meaning. I know it's some kind of bit word, but I couldn't find anywhere, what the bits means. Is there any documentation for this? Can anyone advise me, please? Thank you.

SerafinTech, 2022-08-02:
the status number is a 16bit integer that corresponds to 4 - 4bit integers as follows

```
bits 0-3: unused

bits 4-7: Device status bits
    0000/0 Reserved
    0001/1 Flash update in progress
    0010/2 Reserved
    0011/3 Reserved
    0100/4 Flash is bad
    0101/5 Faulted Modes
    0110/6 Run
    0111/7 Program

bits 8-11: Fault status bits
    0001/1 Recoverable Minor Fault
    0010/2 Unrecoverable Minor Fault
    0100/4 Recoverable Major Fault
    1000/8 Unrecoverable Major Fault
```

currently this library zeros out bits 12-15 in the status value.  If it didn't, those bits would be the key switch position.

This information is in **rslogix designer** software help under **input/output instructions** - **Access the ControllerDevice object**

This is also in the [rslogix5000 controllers general instruction manual](https://literature.rockwellautomation.com/idc/groups/literature/documents/rm/1756-rm003_-en-p.pdf) pg 187

SerafinTech, 2022-08-02:
I'll add some additional properties to the PLC to make it easy to access that information in the next couple days that will allow you to easily tell if the PLC is in run mode.

Thanks for the new feature idea.

petrsysel, 2022-08-03:
Thank you so much! This is exactly what I need!

SerafinTech, 2022-08-03:
FYI The properties are not updated after the initial connection.  To get up to date properties you will need to run

```js
 await  PLC.readControllerProps()
```

closed 2022-10-19 by SerafinTech