SerafinTech/ST-node-ethernet-ip #23: Getting error

Jotan, 2022-08-03:
Hi, I have an error 15 with plc subscription, PLC is a controllogix 5580 my connection to this PLC is:

`const PLC = new Controller(false);`

And the error ir :

`{generalStatusCode: 15, extendedStatus: Array(1) }`

SerafinTech, 2022-08-03:
What firmware version is on the controller?  Can you share a little more code?  Check External Access of the tag you want to read/write to make sure it is set for Read/Write

Jotan, 2022-08-03:
Thank you for the response,  the firmware of the controller is the 30.011, all tags are R/W , 

`const {Controller, Tag,EthernetIP, TagGroup} = require('st-ethernet-ip')
const { DINT, BOOL,BIT_STRING,INT,STRUCT} = EthernetIP.CIP.DataTypes.Types;
const PLC = new Controller(false,{unconnectedSendTimeout: 5064 });
temperaturas = new Tag('TEMPERATURAS_CONTROL','mainProgram',INT,1,0,10)
set_point= new Tag('SET_POINT','mainProgram',INT,1,0,2)
tiempos = new Tag('TIEMPOS','mainProgram',INT,1,0,10)
PLC.subscribe(temperaturas); 
PLC.subscribe(set_point);
PLC.subscribe(tiempos);

PLC.connect("192.168.1.10", 0).then(async () => {
  // Set Scan Rate of Subscription Group to 50 ms (defaults to 200 ms)
 PLC.scan_rate = 200;

  PLC.scan();
  

 
  
}).catch(async e => {

});
`

SerafinTech, 2022-08-04:
I checked your code on the exact same PLC and firmware and had no errors.  Check that the arrays you are reading are actually the length specified.  Check the security properties on the PLC in Logix Designer to make sure there isn't some security setup or communication isn't restricted on the port.

Jotan, 2022-08-05:
I have like 50 tag subscription and the error code is 15, that means that i have error too much data, if i generate tag group, do you think that is better?, or i'm thinking doing a double connection with different tags subs.

Jotan, 2022-08-06:
I think is solved, there is an error when  i have tags with arrays, but when i have tag with one values, i don't that error anymore.