SerafinTech/ST-node-ethernet-ip #27: Error reading predefined ALARM tags

srebling, 2022-11-11:
Can not read a predefined tag of type ALARM or ALARM_DIGITAL.

## Current Behavior

The readTag() function returns the following error:

`{ generalStatusCode: 15, extendedStatus: [] }`

## Expected Behavior

Read the tag without error. 

## Possible Solution (Optional)

Might have something to do with the extra SINT members that come along with structures that have BOOL members (one SINT for every 8 BOOLs). Just a thought because I am also experiencing incorrect data being written to DINT members, if the structure contains any BOOL members.

## Context

## Steps to Reproduce (for bugs only)

Run this code with a tag named 'ALM_1' defined as type ALARM.
```
const {Controller} = require('st-ethernet-ip')

const PLC = new Controller()

PLC.connect('10.0.0.250',0).then(async () => {

  console.log('Connected...')
  const udt = PLC.newTag('ALM_1')
  
  await PLC.readTag(udt)
  console.log('ALM_1:', udt.value)

}).catch(e => {
  console.log('Error: ', e)
})
```

## Your Environment

* Package version (Use `npm list` - e.g. *1.0.6*): 2.5.0
* Node Version (Use `node --version` - e.g. *9.8.0*): 16.14.2
* Operating System and version: OSX 10.15.7
* Controller Type (eg 1756-L83E/B): 1769-L24ER-QB1B CompactLogix 5370
* Controller Firmware (eg 30.11): 30.011

SerafinTech, 2022-11-11:
I read and wrote an ALARM type tag using the same code structure as you and did not have an issue on a 1769-L32E. Double check that it is a controller scope tag and the name is correct. Check external access is also Read/Write.

srebling, 2022-11-12:
Actually, reading type ALARM does work for me. It's ALARM_DIGITAL and ALARM_ANALOG types that do not. I noticed the External Access field is grayed-out in the controller tags editor for those types. Can not be changed from Read/Write. Not sure why that is or if it matters, but it is set to Read/Write.

SerafinTech, 2022-11-12:
Looks like Rockwell blocks reading the whole structure of ALARM_DIGITAL and ALARM_ANALOG, but you can read the individual components of the structure with separate tags for each value. The ALARM_DIGITAL and ALARM_ANALOG  types and corresponding instructions ALMD and ALMA are designed to interact with the Factory Talk Alarm and Event System.

srebling, 2022-11-13:
As a work around I just copied the data I need from the alarm tag into a new structure in the PLC and read that. Want to stick with UDT's instead of separate tags. Works nicely. Thanks

closed 2022-11-13 by srebling