SerafinTech/ST-node-ethernet-ip #28: ControllerManager 'TagChanged' event anomaly

srebling, 2022-11-18:
When monitoring a UDT with BOOL members, the first member's value unexpectedly changes if there is a subsequent BOOL member in the UDT.
## Current Behavior

Value of the first member of the UDT unexpectedly changes. Could create unpredictable behavior of the control system.
## Expected Behavior

Monitor UDT's with BOOL members without effecting the first member.
## Possible Solution (Optional)

## Context

This use case only needs to read values, never write.
## Steps to Reproduce (for bugs only)

1. Create a UDT (UDT_1) in the PLC with two members. The first as a DINT, the second a BOOL.  
2. Run the code below and monitor the value of the DINT member (RSLogix).
3. Change the value of the DINT and it will change back to something else (depending on the value of the BOOL).
4. Change the value of the BOOL and the DINT value will unexpectedly change.

```
const {ControllerManager} = require('st-ethernet-ip')

const cm = new ControllerManager();

//addController(ipAddress, slot = 0, rpi = 100, connected = true, retrySP = 3000, opts = {})
const cont = cm.addController('10.0.0.250');

cont.connect();

//addTag(tagname, program = null, arrayDims = 0, arraySize = 0x01)
cont.addTag('UDT_1')

cont.on('TagChanged', (tag, prevValue) => {
  console.log(tag.name, ' changed from ', prevValue, ' => ', tag.value)
})

cont.on('error', (e) => {
  console.log(e)
})
```

## Your Environment

* Package version (Use `npm list` - e.g. *1.0.6*): 2.5.0
* Node Version (Use `node --version` - e.g. *9.8.0*): 16.14.2
* Operating System and version: OSX 10.15.7
* Controller Type (eg 1756-L83E/B): 1769-L24ER-QB1B CompactLogix 5370
* Controller Firmware (eg 30.11): 30.011

SerafinTech, 2022-11-21:
I was able to reproduce your error.  I will need to dig into this.

SerafinTech, 2022-11-21:
Found the issue and have a fix.  Testing it out and should be pushed out by tonight.

srebling, 2022-11-21:
Thank you. Also see something similar with the writeTag() function.

SerafinTech, 2022-11-22:
WriteTag function should be fixed too.  The code that converted the JSON tag value object from BOOL to buffer data was completely wrong. 2c30c15ff67b49af1f23b57b0454d87031cc1e8c

srebling, 2022-11-28:
Working as expected now. Thank you!

In my sample code above, are there any writes being performed behind the scenes by the ControllerManager object? The fact that tag values changed without commanding a write makes me nervous.

SerafinTech, 2022-11-29:
So what happened was that the tag structure object was incorrectly converted to buffer data. When the Nodejs program saw that the buffer data didn't match the buffer data being read from the PLC, the program 'thought' that there was updated data that needed to be written to the PLC controller.

closed 2023-02-07 by SerafinTech