SerafinTech/ST-node-ethernet-ip #31: STRING type returning Buffer data only

skykep, 2023-02-28:
Maybe I'm jumping the gun, but I thought the support for STRING data types was added; however, upon trying to read them, I'm only able to get Buffer data.

 value: <Buffer 08 00 00 00 30 31 2d 30 33 2d 32 33 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... 38 more bytes>

## Current Behavior
Returns buffer data instead of string text

## Expected Behavior
Return string text

## Your Environment
* Package version (Use `npm list` - e.g. *1.0.6*): st-ethernet-ip@2.5.2
* Node Version (Use `node --version` - e.g. *9.8.0*): 12.18
* Operating System and version: Ubuntu 18.04
* Controller Type (eg 1756-L83E/B): 5069-L306ER
* Controller Firmware (eg 30.11): 30

Iterastor, 2023-04-12:
if you declare the tag inside de PLC conexion it returns a String

example

const PLC = new controller ();
PLC.connect('192.168.0.1',0).then(async() => {

//newTag
const array_string= PLC.newTag('A9', null, true, 1);
//Get the length of array string
await PLC.getTagArraySize(calidad);

//Reading array_string from PLC
await PLC.readTag(array_string);
console.log(array_string);
});

SerafinTech, 2023-04-20:
There was an issue with arrays of structures/UDTs .  Try installing latest version to see if it fixes your issue.

closed 2023-08-09 by SerafinTech