SerafinTech/ST-node-ethernet-ip #33: Add typescript declaration file

bahodge, 2023-04-20:
## Current Behavior

Typescript is unsupported as there is no `@types/st-ethernet-ip` which makes it difficult to work with typescript. 

## Expected Behavior

I should be able to either install this lib or install `@types/st-ethernet-ip` and get type declarations.

## Possible Solution (Optional)

- add `index.d.ts` to define all public types. 

## Context

My team is beginning to adopt this library as one of our core libs and using it is "janky" because we have to comb through source code to figure out what everything does. Adding a declaration file along with some basic documentation will make working with this library much easier.

## Steps to Reproduce (for bugs only)

## Your Environment
* Package version `st-ethernet-ip@2.5.2`
* Node Version `18.12.0`
* Operating System and version: Ubuntu 20.04.6 LTS x86_64
* Controller Type: N/A
* Controller Firmware: N/A

SerafinTech, 2023-04-20:
I'll add this to my todo list.

SerafinTech, 2023-05-04:
Added support in merge #34

bahodge, 2024-04-11:
At least in the version we are using, this requirement has been satisfied. I think we can close this issue now.

closed 2024-11-27 by bahodge