SerafinTech/ST-node-ethernet-ip #38: Code example to retrieve all tags and values

SerafinTech, 2023-08-08:
Code to retrieve all tags and values.  Works only with newest version.

```javascript
const eip = require('st-ethernet-ip')

let cont = new eip.Controller()

cont.connect(IP_ADDRESS)
.then(async () => {
    let tagGroup = new eip.TagGroup()
    
    for (let i = 0; i < cont.tagList.length; i++) {
        let tag = cont.newTag(cont.tagList[i].name, cont.tagList[i].program, false, cont.tagList[i].type.arrayDims)

        tagGroup.add(tag)

        if (cont.tagList[i].type.arrayDims > 0) { 
            await cont.getTagArraySize(tag)
        }  
    }

    await cont.readTagGroup(tagGroup)
    tagGroup.forEach(t => {
        console.log(t.name, t.value)
    })
})
```

bachphi, 2023-11-21:
I am using node v20.9 with st-ethernet-ip v2.67.  The above example give error:

           throw new Error("Tagname Must be of Type <string>");
                  ^

Error: Tagname Must be of Type <string>
    at new Tag (/home/pitech/node_modules/st-ethernet-ip/dist/tag/index.js:54:19)
    at new Structure (/home/pitech/node_modules/st-ethernet-ip/dist/structure/index.js:28:9)
    at Controller.newTag (/home/pitech/node_modules/st-ethernet-ip/dist/controller/index.js:1209:19)
    at /home/pitech/demo2.js:10:24
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)

Node.js v20.9.0

bachphi, 2023-11-21:
11/21/23  Tried it on a PC with node 21.2 and st-etheret-ip v2.67
still get the same error

C:\MACH\NODES\node_modules\st-ethernet-ip\dist\tag\index.js:54
            throw new Error("Tagname Must be of Type <string>");
                  ^

Error: Tagname Must be of Type <string>
    at new Tag (C:\MACH\NODES\node_modules\st-ethernet-ip\dist\tag\index.js:54:19)
    at Controller.newTag (C:\MACH\NODES\node_modules\st-ethernet-ip\dist\controller\index.js:1215:19)
    at C:\MACH\NODES\NodeJS\GetAllTags.js:11:24
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)

Node.js v21.2.0

SerafinTech, 2023-11-22:
Out of town till Monday.  I'll check this out after the holiday.