SerafinTech/ST-node-ethernet-ip #42: Expose handy classes and properties!

flacohenao, 2023-08-18:
Hey, I'm giving a try to your lib, and I found that for some specific cases, the `extController` class, can be used at will without having to use the `ControllerManager` class.. this will be handy for example when we want to create just one controller and attach some custom information to it in an outer scope.

as you are exporting the `ControllerManager` by default, you are being forced to instatiate always the manager to get the extController which is pointless.

Also, I noted that the `extController` has a great support for auto-reconnection, but it will be nice to also expose the capability to desable the auto-reconnection, which right now we can not change or initialize at building time of the `extController`

It would be awesome if we can do it.

flacohenao, 2023-08-18:
Here, this is the problem with the `autoReconnect `feature: 

You are always setting it to true, no matter the value passed

[image]

SerafinTech, 2023-08-20:
Sounds like a good idea see commits:
c18d594039b82ba0b2d5792bf8e50619ed94749d
4dd8acb9a9d7e307c37a90644b805c2a6fcf1378