SerafinTech/ST-node-ethernet-ip #47: Support for Micro800

TeeLusk, 2023-08-25:
## Current Behavior

Would love for this package to support Micro800 controllers. Any attempt at a connection returns
```
{ generalStatusCode: 1, extendedStatus: [ 785 ] }
```
 I've found a similar package that works with Micro800, seemingly with some small changes from other PLCs (See [node-logix](https://github.com/msvargas/node-logix/blob/202c7461169b78a6f12b51b7d6fc69ff8fb88b61/src/eip-socket.ts#L93)) but it isn't actively maintained and doesn't have support for some of the nice things built into this package like TagChanged or ControllerManager. 
I'm new to this library and PLC communication in general but even if I could get a couple pointers on where these changes would go I would make a PR and add it. 

## Expected Behavior

## Possible Solution (Optional)

## Context

## Steps to Reproduce (for bugs only)

1.
2.
3.
4.

## Your Environment

* Package version (Use `npm list` - e.g. *1.0.6*):
* Node Version (Use `node --version` - e.g. *9.8.0*):
* Operating System and version:
* Controller Type (eg 1756-L83E/B):
* Controller Firmware (eg 30.11):

SerafinTech, 2023-08-25:
try Buffer.from([]) where you enter the slot number. This creates an empty path for simple devices.

```javascript
const {ControllerManager} = require('st-ethernet-ip')

const cm = new ControllerManager();

//addController(ipAddress, slot = 0, rpi = 100, connected = true, retrySP = 3000, opts = {})
const cont = cm.addController('192.168.86.200', Buffer.from([]));

cont.connect();

```
If that doesn't work, then also try unconnected messaging.

```javascript
const {ControllerManager} = require('st-ethernet-ip')

const cm = new ControllerManager();

//addController(ipAddress, slot = 0, rpi = 100, connected = true, retrySP = 3000, opts = {})
const cont = cm.addController('192.168.86.200', Buffer.from([]), 100, false);

cont.connect();

```

I don't have a Micro800 to test.  If someone let me borrow one, I'm sure I could get it to work.

TeeLusk, 2023-08-25:
First option connected! I closed the issue when I saw a recent commit for adding Micro800 support and was figuring out how to pass the buffer, appreciate the quick response.

TeeLusk, 2023-08-25:
I am now having some issues now with the subscription/scan methods. The following works and I can verify it is setting the Tag value in Connected Components Workbench, but when I uncomment the PLC.scan() method, with or without the other writeTag code I get an unhandled rejection error

Code:
``` javascript
const { Controller, Tag, EthernetIP } = require("st-ethernet-ip");
const { BOOL } = EthernetIP.CIP.DataTypes.Types;

const PLC = new Controller();

let tag1 = new Tag("TEST_TAG_1", null, BOOL);
PLC.subscribe(tag1);

PLC.connect("192.168.1.10", Buffer.from([]))
  .then(async () => {
    tag1.value = true;
    PLC.writeTag(tag1);

    // PLC.scan_rate = 50;
    // PLC.scan();
  })
  .catch((err) => {
    console.log(err);
  });
```

Error:
``` 
node:internal/process/promises:288
            triggerUncaughtException(err, true /* fromPromise */);
            ^

[UnhandledPromiseRejection: This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason "#<Object>".] {
  code: 'ERR_UNHANDLED_REJECTION'
}
```

Any thoughts what might be causing that?