SerafinTech/ST-node-ethernet-ip #48: Micro800 Controller.scan() Error

TeeLusk, 2023-08-28:
## Current Behavior
I am now having some issues now with the subscription/scan methods. The following works and I can verify it is setting the Tag value in Connected Components Workbench, but when I uncomment the PLC.scan() method, with or without the other writeTag code I get an unhandled rejection error

Code:
``` javascript
const { Controller, Tag, EthernetIP } = require("st-ethernet-ip");
const { BOOL } = EthernetIP.CIP.DataTypes.Types;

const PLC = new Controller();

let tag1 = new Tag("TEST_TAG_1", null, BOOL);
PLC.subscribe(tag1);

PLC.connect("192.168.1.10", Buffer.from([]))
  .then(async () => {
    tag1.value = true;
    PLC.writeTag(tag1);

    // PLC.scan_rate = 50;
    // PLC.scan();
  })
  .catch((err) => {
    console.log(err);
  });
```

Error:
``` 
node:internal/process/promises:288
            triggerUncaughtException(err, true /* fromPromise */);
            ^

[UnhandledPromiseRejection: This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason "#<Object>".] {
  code: 'ERR_UNHANDLED_REJECTION'
}
```

Any thoughts what might be causing that?

SerafinTech, 2023-08-28:
I tried your code on a non Micro800 without a problem.

For the ```PLC.scan()```

Try:

```javascript
PLC.scan().catch(e => console.log(e))
```

To catch the scan error

TeeLusk, 2023-08-28:
```
{ generalStatusCode: 8, extendedStatus: [] }
```

SerafinTech, 2023-08-28:
That is a service not supported error.  The scan function uses MULTIPLE_SERVICE_PACKET service to send multiple read or write requests at the same time. It appears that the Micro800 doesn't support this service.

TeeLusk, 2023-08-29:
@SerafinTech What CompactLogix model(s) are you using to develop this library? We've also got a Micro820 that we'd be willing to send over to see if we can get some sort of scan() equivalent supported on that platform.

anthonywebb, 2023-08-29:
@SerafinTech if the library can read/write to the micro800 no problem, do you see any issues to just put a read inside a polling loop and roll your own "scanner" via hitting the PLC say every second?  Is that essentially what the library is doing at its core anyway?  Polling is usually less than ideal.  Also, I'd be curious to know which PLC you are using and have tested against.  Perhaps a compatibility chart and feature matrix would be a nice addition.  Let me know if youd like to get one going I am happy to help in any way I can.

SerafinTech, 2023-08-29:
@TeeLusk I currently develop and test against a CompactLogix 1769-L32E Firmware 20.14 . I wish I had a bunch of controllers to test against.  ControlLogix and CompactLogix "should" all be compatible with the library. I would be happy to create a scan function for the Micro820 if you sent me the controller, but I couldn't promise a timeline for completing.  It shouldn't be long, but you never know what problems come up.

SerafinTech, 2023-08-29:
@anthonywebb You can do polling which is what the scan function basically does, but you have to wait for the read or write to complete before you initiate another read or write and catching errors can be tricky.

anthonywebb, 2023-08-30:
In general I am finding that maybe the micro800 just isn’t cut out for scanning, which is a shame because it is a great entry point into PLC. 

[image]
[image]