SerafinTech/ST-node-ethernet-ip #53: I can't write to a tag, that does reads successfully.

William-Brumble, 2023-11-23:
## Summary:
I want to write to some indexes inside an array in a program scope tag.
I'm able to read the tags, but when I attempt to write I get the following error:

```text
Error: TIMEOUT occurred while writing Writing Tag: Program:[program].[tag-name]
```

<i>Do you guys have a discord server?</i>

## Current Behavior:
When attempting to write to a tag array it's rejecting the promise on a TIMEOUT error.

## Expected Behavior:
Write to manipulated tag array indexes.

## Context
```typescript
export const sendCommand = async (command: Command) => {
    try {
        const PLC = new Controller();
        await PLC.connect(command.ip, command.slot);

        const programName = command.commandTag.split(":")[1].split(".")[0];
        const tagName = command.commandTag.split(".")[1];
        const tagObj = new Tag(tagName, programName, Types.REAL, 0, 1, 1000);

        await PLC.readTag(tagObj);

        for (let i = 0; i < command.parameters.length; i++) {
            const parameter = command.parameters[i];
            tagObj.value[i] = parameter.value;
        }
        tagObj.value[command.offsetNum] = command.command;

        await PLC.writeTag(tagObj); // rejects this promise

        await PLC.disconnect();
    } catch (error) {
        console.log(error);
    }
};
````

## Environment:
* Package version (Use `npm list` - e.g. *1.0.6*):  <b>2.6.7</b>
* Node Version (Use `node --version` - e.g. *9.8.0*):  <b>18.16.1</b>
* Operating System and version:  <b>Ubuntu 23.04</b>
* Controller Type (eg 1756-L83E/B):  <b>1756-L84E</b>
* Controller Firmware (eg 30.11): <b>32.013</b>

William-Brumble, 2023-12-16:
To update this issue, it was being caused by the large array, 1k elements. When I lowered the number of elements the timeouts went away.

SerafinTech, 2023-12-18:
Yep. That is a bug / missing implementation.  Latest commit should work.  pull from repo and try out. Not published to NPM yet.
You will have to run `npm run build` where it installed to compile the Typescript.