SerafinTech/ST-node-ethernet-ip #56: Controller Manager not auto-reconnecting

skykep, 2024-01-17:
Controller Manager does not auto-reconnect to a PLC.  

## Current Behavior

After a download to a PLC, tags are no longer read from that PLC.

## Expected Behavior

Tags should resume polling from that PLC.

## Possible Solution (Optional)

## Context

## Steps to Reproduce (for bugs only)

1.
2.
3.
4.

## Your Environment

* Package version (Use `npm list` - e.g. *1.0.6*): 2.6.7
* Node Version (Use `node --version` - e.g. *9.8.0*):
* Operating System and version:
* Controller Type (eg 1756-L83E/B): 1756-L63
* Controller Firmware (eg 30.11): 16

skykep, 2024-01-18:
My temporary workaround:

setInterval(async function () {
  let tagValues = cm.getAllValues();
  let heartbeat = cont.tags.find(o => o.tagname === 'Heartbeat').tag.state.tag;
  console.log(heartbeat.value);
  if (heartbeat.value == false) {
    console.log("Restarting connection...");
    cont.connect();
  }
  heartbeat.value = false;
}, 5000);

In the PLC, I just have a single rung for 'OTE(Heartbeat);'

SerafinTech, 2024-01-20:
I was able to duplicate the issue. After power cycling it will reconnect, but after download it doesn't.  I will see if I can figure out a fix.

skykep, 2024-01-21:
Thanks Jason.  Glad I was able to find a way to break it.   Haha

SerafinTech, 2024-01-27:
New version should fix

skykep, 2024-01-27:
Thanks Jason.  I'll give it a shot tomorrow.