SerafinTech/ST-node-ethernet-ip #67: readTag() throws exception reading BOOL array tag elements such as BoolTag[2]

ryangriggs, 2024-06-11:
## Current Behavior
Controller: 1769-L35E, firmware 20.19
Controller-scoped boolean array tag: BoolTag, type BOOL[32]

Attempting to read BoolTag[2] throws exception: {generalStatusCode: 5, extendedStatus: Array(0)}
Attempting to read BoolTag[0] returns an array of all array elements (32 boolean values).

Code:
```
const tag1 = new Tag("BoolTag[0], null, BOOL);
const tag2 = new Tag("BoolTag[2], null, BOOL);

plc.readTag(tag1)
.then(() => {
    console.log(`Tag value: ${tag1.value} of type ${typeof tag1.value}`);
   // Output: Tag value: false,true,false,true,false,true,false,true,false,true,false,false,false,false,false,false,false,false,false,false,false,false,false,false,false,false,false,false,false,false,false,false of type object
})
.catch((err) => {
    console.error(err);
});

plc.readTag(tag2)
.then(() => {
    console.log(`Tag value: ${tag2.value} of type ${typeof tag2.value}`);
})
.catch((err) => {
    console.error(err);
   // Output: {generalStatusCode: 5, extendedStatus: Array(0)}
});
```

Reading INT array tags works with no issues (i.e. "IntTag[2]" returns the tag integer value)

## Expected Behavior
Expect that any attempt to read a BOOL tag array element returns a single BOOL value, not an array of values, and not throws an exception.

## Possible Solution (Optional)
n/a

## Context
Trying to read a few individual BOOL array tags for status values, without reading the entire array, which could be many hundreds of elements.

## Steps to Reproduce (for bugs only)
See code above.

## Your Environment
* Package version (Use `npm list` - e.g. *1.0.6*):  st-ethernet-ip@2.7.2
* Node Version (Use `node --version` - e.g. *9.8.0*): v21.7.1
* Operating System and version: Windows 10 x64 22H2
* Controller Type (eg 1756-L83E/B): 1769-L35E
* Controller Firmware (eg 30.11): 20.19

SerafinTech, 2024-06-12:
Yep. I duplicated that bug.  I'll see what's going on

SerafinTech, 2024-06-12:
I believe Arrays of type BOOL is stored as an integer.  So to access that, use a '.' instead of '[ ]'.

change BoolTag[0] to BoolTag.0 and BoolTag[3] to BoolTag.3

ryangriggs, 2024-06-12:
@SerafinTech Thanks for your fast response! It definitely works to access the bits using the dot format for both reading and writing.  I'm curious why it doesn't work with array indices same as other array types?  also, does it support multi-dimensional boolean arrays, and if so, how would you address them?  Would it be BoolArray[0].5 or BoolArray.0.5?

Again thanks for the fast response!

closed 2024-06-12 by ryangriggs

ryangriggs, 2024-06-14:
Apparently I was wrong about writing - I can't seem to get it to write to either boolean array bits or integer bits.  It throws a blank exception on any attempt.