SerafinTech/ST-node-ethernet-ip #71: Write to a tag using ControllerManager?

skykep, 2024-07-14:
## Current Behavior
Can read tags fine.

## Expected Behavior
Write to a tag using ControllerManager

skykep, 2024-08-18:
Still wondering the best way to write data back to the PLC using ControllerManager...

While I wait, I will say that the ControllerManager works great.  I just performed a test on a 1756-L61 I had in slot 0.  I removed it from the chassis, inserted a 1756-L81E, downloaded the same program to the L81, and the ControllerManager picked right back up and the script kept running.  Completely hands-off!

SerafinTech, 2025-04-18:
Here is some example code for implementing the ControllerManager

```js
const {ControllerManager} = require('st-ethernet-ip');

let cm = new ControllerManager();

//If you want to do something with the value every 1000ms , I would set the rpi to something smaller (500) so your sure to have a new value.
cm.addController('192.168.121.10', 0, 100);
//Always connect controller after adding it
cm.controllers[cm.controllers.length - 1].connect();
//Add Your tags to the controller (Example is DINT Controller scope tag called 'RATE_TEST99' )
cm.controllers[cm.controllers.length - 1].addTag('RATE_TEST99');
//Add Your tags to the controller (Example is REAL Program:MainProgram scope tag called 'TEST_POINT1' )
cm.controllers[cm.controllers.length - 1].addTag('TEST_POINT1', 'MainProgram');
cm.controllers[cm.controllers.length - 1].addTag('testUDT3[0]', 'MainProgram');

// READING TAG VALUE EXAMPLE
//Do something with tag values every 1000ms
setInterval(() => {
  let tagValues = cm.getAllValues();

  // Display all tag names and values at controller '192.168.121.10'
  console.log(tagValues['192.168.121.10']);
  // Display tag named 'RATE_TEST99' value
  console.log(tagValues['192.168.121.10']['RATE_TEST99']);
  // Display tag named 'TEST_POINT1' value
  console.log(tagValues['192.168.121.10']['Program:MainProgram.testUDT3[0]']);

  //Another way to get a current tag value
  console.log(cm.controllers.find(c => c.ipAddress === '192.168.121.10').tags.find(n => n.tagname === 'RATE_TEST99').tag.value);

}, 1000);

//WRITING TAG VALUE EXAMPLE
//Write a incrementing value to a controller tag after 1000ms
setInterval(() => {
  //Increment value and write to DINT controller scope tag 'RATE_TEST99' at controller '192.168.121.10'
  let tag1 = cm.controllers.find(c => c.ipAddress === '192.168.121.10').tags.find(n => n.tagname === 'RATE_TEST99').tag;
  tag1.value = tag1.value + 1;

  //Increment value and write to REAL Program:MainProgram scope tag 'TEST_POINT1' at controller '192.168.121.10' 
  let tag2 = cm.controllers.find(c => c.ipAddress === '192.168.121.10').tags.find(n => (n.tagname === 'testUDT3[0]' && n.program === 'MainProgram')).tag;
  tag2.value.UDT1[0].INT1 = tag2.value.UDT1[0].INT1 - 1000

  //ALWAYS find the tag as shown before doing anything with the tag to insure you are not using a stale tag instance

}, 1000);
```