SerafinTech/ST-node-ethernet-ip #75: Support for unsigned data types? (UINT, UDINT, etc.)

skykep, 2025-01-08:
Is there any plan to add support for the unsigned data types that were introduced in v32 of Logix firmware?

SerafinTech, 2025-04-19:
I currently do not have a controller that supports unsigned data types (I test on a CompactLogix 5370). Unsigned support will have to wait until I get my hands on a different controller.

I'll look into the BOOL array.

skykep, 2025-04-21:
> I currently do not have a controller that supports unsigned data types (I test on a CompactLogix 5370). Unsigned support will have to wait until I get my hands on a different controller.
> 
> I'll look into the BOOL array.

How can I contact you?  I have an option that may help you test on a 5380 controller.