SerafinTech/ST-node-ethernet-ip #77: Disconnect Option?

lbhopper, 2025-03-06:
Is there currently a disconnect option for this? I'm using process.exit() to exit out of my function instead of waiting for a timeout:
```
async function readPLC() {
    try {
        await PLC.connect('192.168.0.1, 0);

        const tagData = {};

        for (const tagName of tagArray) {
            const tag = new Tag(tagName);
            await PLC.readTag(tag);
            tagData[tagName] = tag.value;
        }

        console.log(JSON.stringify(tagData, null, 2));

        process.exit(0);
    } catch (err) {
        console.error(JSON.stringify({ error: err.message }, null, 2));
        process.exit(1);
    }
}

readPLC();
```

lbhopper, 2025-03-26:
So there is a disconnect option, but I'm assuming it wasn't shown in the code examples because it's not functioning properly. I've had to do multiple tests to figure this out, but here is my code:

PLC.disconnect(); closes connection to the target PLC (verified on a CompactLogix 5069-L310ER/A) but doesn't clean up the timers left behind which is why the script hangs for roughly 30 secs before exiting. You can see how many timers are called using this function before connecting to the PLC:
```
const activeTimeouts = new Set();

const realSetTimeout = global.setTimeout;
global.setTimeout = function (fn, delay, ...args) {
    console.log(`setTimeout called with delay: ${delay}`);
    const timeoutId = realSetTimeout(fn, delay, ...args);
    activeTimeouts.add(timeoutId);
    return timeoutId;
};
```
Output:
```
setTimeout called with delay: 10000
setTimeout called with delay: 10000
setTimeout called with delay: 10000
setTimeout called with delay: 10000
setTimeout called with delay: 40000
...
```

To actually clear the timers, set this function:
```
const realClearTimeout = global.clearTimeout;
global.clearTimeout = function (id) {
    activeTimeouts.delete(id);
    realClearTimeout(id);
};
```
Then call the function after PLC.disconnect()
```
console.log(`Clearing ${activeTimeouts.size} timers...`);
for (const timeoutId of activeTimeouts) {
    clearTimeout(timeoutId);
}
```
To display active timeouts:
```
console.log("Active Timeout IDs:", [...activeTimeouts]);
```

Full code:
```
const { Controller, Tag } = require('st-ethernet-ip');

const PLC = new Controller();
const tagArray = ["Main_E_Stop_OK"];

const activeTimeouts = new Set();

const realSetTimeout = global.setTimeout;
global.setTimeout = function (fn, delay, ...args) {
    console.log(`setTimeout called with delay: ${delay}`);
    const timeoutId = realSetTimeout(fn, delay, ...args);
    activeTimeouts.add(timeoutId);
    return timeoutId;
};

const realClearTimeout = global.clearTimeout;
global.clearTimeout = function (id) {
    activeTimeouts.delete(id);
    realClearTimeout(id);
};

async function readPLC(ipAddress) {
    try {
        await PLC.connect(ipAddress, 0);
        console.log("Connected to PLC");

        const tagData = {};

        for (const tagName of tagArray) {
            const tag = new Tag(tagName);
            await PLC.readTag(tag);
            tagData[tagName] = tag.value;
        }

        console.log("PLC data:");
        console.log(JSON.stringify(tagData, null, 2));
    } catch (error) {
        console.error("Error during PLC operation:", error);
    } finally {
        try {
            console.log("Calling PLC.disconnect...");

            await PLC.disconnect();

            console.log("Disconnected successfully");

            console.log("Active Timeout IDs:", [...activeTimeouts]);

            console.log(`Clearing ${activeTimeouts.size} timers...`);
            for (const timeoutId of activeTimeouts) {
                clearTimeout(timeoutId);
            }
            activeTimeouts.clear();

        } catch (disconnectError) {
            console.error("Error while disconnecting:", disconnectError);
        }
    }
}

readPLC('192.168.0.200');

```

lbhopper, 2025-03-26:
UPDATE:

Was able to correct the open timers with the following:
../st-ethernet-ip/dist/enip/index.js > line 159
```
await promiseTimeout(
    new Promise((resolve, reject) => {
        let options = {
            port: config_1.EIP_PORT,
            host: IP_ADDR,
            localAddress: localAddress
        };

        let socket = super.connect(options, () => {
            this.state.TCP.establishing = false;
            this.state.TCP.established = true;
            this.write(registerSession());
            resolve();
        });

        socket.on("error", (err) => {
            reject(new Error("SOCKET error: " + err.message));
        });

    }),
    timeoutSP,
    connectErr
).catch((err) => {
    console.error("Promise timed out or failed:", err);
    if (socket) {
        socket.destroy();
        console.log("Socket forcefully destroyed due to timeout.");
    }
});
```

../st-ethernet-ip/dist/utilities/index.js > promiseTimeout:
```
const timeouts = new Set();

const promiseTimeout = (promise, ms, error = new Error("ASYNC Function Call Timed Out!!!")) => {
    return new Promise((resolve, reject) => {
        const timeoutId = setTimeout(() => {
            timeouts.delete(timeoutId);
            reject(error);
        }, ms);

        timeouts.add(timeoutId);

        promise.then((res) => {
            clearTimeout(timeoutId);
            timeouts.delete(timeoutId);
            resolve(res);
        }).catch((err) => {
            clearTimeout(timeoutId);
            timeouts.delete(timeoutId);
            reject(err);
        });
    });
};
exports.promiseTimeout = promiseTimeout;
exports.timeouts = timeouts;
```

../st-ethernet-ip/dist/controller/index.js > async disconnect:
```
async disconnect() {
    const timeouts = utilities_1.timeouts;
    if (super.established_conn === true) {
        const closeid = await this.forwardClose();
        if (!closeid)
            throw new Error("Failed to End Connected EIP Session with Forward Close Request");
    }

    // Clear all stored timeouts
    for (const timeoutId of timeouts) {
        clearTimeout(timeoutId);
    }
    timeouts.clear();

    super.destroy();
    this._removeControllerEventHandlers();
    return "disconnected";
}
```

Now the script will clean up the timers when calling PLC.disconnect();

```
const { Controller, Tag } = require('st-ethernet-ip');

const PLC = new Controller();
const tagArray = ["Main_E_Stop_OK", "Bagger1_Estop_OK", "Bagger2_Estop_OK", "Bagger3_Estop_OK"];

async function readPLC(ipAddress) {
    try {
        await PLC.connect(ipAddress, 0);
        console.log("Connected to PLC");

        const tagData = {};

        for (const tagName of tagArray) {
            const tag = new Tag(tagName);
            await PLC.readTag(tag);
            tagData[tagName] = tag.value;
        }

        console.log(JSON.stringify(tagData, null, 2));

    } catch (error) {

        console.error("Error during PLC operation:", error);

    } finally {
        
        try {
            await PLC.disconnect();
        } catch (disconnectError) {
            console.error("Error while disconnecting:", disconnectError);
        }
    }
}

readPLC('192.168.0.100');
```

alex-controlx, 2025-10-04:
I'm migrating this library to a cross-JS codebase and to TypeScript. 

It was solved by modifying the utilities file (line 18):
https://github.com/controlx-io/ethernet-ip-controller/blame/cab2fb23f3e9b350c44c3870b89eaef307654f84/src/utils/utilities.ts#L18